techman83/Filament-Scale-Enhanced #10: Missing dependencies RPi.GPIO and Mock

ali1234, 2021-05-03:
When attempting to install plugin using the plugin manager on octopi 0.18 stable the following error happens:

```
2021-05-03 20:45:54,303   /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp_0eg9bt7/Filament_Scale_Enhanced.zip --no-cache-dir
2021-05-03 20:45:56,801 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2021-05-03 20:45:56,802 > Processing /tmp/tmp_0eg9bt7/Filament_Scale_Enhanced.zip
2021-05-03 20:45:58,780 ! ERROR: Command errored out with exit status 1:
2021-05-03 20:45:58,781 ! command: /home/pi/oprint/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-hlp1on8u/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-hlp1on8u/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-iz4lm_4h
2021-05-03 20:45:58,781 ! cwd: /tmp/pip-req-build-hlp1on8u/
2021-05-03 20:45:58,781 ! Complete output (16 lines):
2021-05-03 20:45:58,782 ! Traceback (most recent call last):
2021-05-03 20:45:58,782 ! File "/tmp/pip-req-build-hlp1on8u/filament_scale_enhanced/hx711.py", line 5, in <module>
2021-05-03 20:45:58,782 ! import RPi.GPIO as GPIO
2021-05-03 20:45:58,782 ! ModuleNotFoundError: No module named 'RPi'
2021-05-03 20:45:58,783 ! 
2021-05-03 20:45:58,783 ! During handling of the above exception, another exception occurred:
2021-05-03 20:45:58,783 ! 
2021-05-03 20:45:58,784 ! Traceback (most recent call last):
2021-05-03 20:45:58,784 ! File "<string>", line 1, in <module>
2021-05-03 20:45:58,784 ! File "/tmp/pip-req-build-hlp1on8u/setup.py", line 5, in <module>
2021-05-03 20:45:58,785 ! from filament_scale_enhanced import __version__
2021-05-03 20:45:58,785 ! File "/tmp/pip-req-build-hlp1on8u/filament_scale_enhanced/__init__.py", line 7, in <module>
2021-05-03 20:45:58,785 ! from .hx711 import HX711
2021-05-03 20:45:58,785 ! File "/tmp/pip-req-build-hlp1on8u/filament_scale_enhanced/hx711.py", line 7, in <module>
2021-05-03 20:45:58,786 ! import Mock.GPIO as GPIO
2021-05-03 20:45:58,786 ! ModuleNotFoundError: No module named 'Mock'
2021-05-03 20:45:58,786 ! ----------------------------------------
2021-05-03 20:45:58,786 ! ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

techman83, 2021-05-04:
I know what's going on here. Issue was hidden from me as I have another plugin that depends on RPi.GPIO. I'll come up with something and release a fix today.

techman83, 2021-05-04:
I've released a fix, v0.2.10. Thanks for reporting!

commit a8a978ed (2021-05-04): fix: Dependency errors during setup

Open version file directly to avoid requiring runtime dependencies during setup - fixes #10
PR #11 (2021-05-04): Fix Setup
## Adds
- Run `python setup.py --version`, will fail if similar issues to #10 arise

## Fixes
- Remove direct import from `setup.py`, fixes #10
closed 2021-05-04 by techman83