techman83/Filament-Scale-Enhanced #12: Error: No filament sensor detected.

ali1234, 2021-05-04:
The plugin can be installed now but it still doesn't work.

It works with Victor's version, and with the original standalone hx711py library.

ali1234, 2021-05-04:
Forgot to mention. It does this on the settings screen and prevents me from calibrating.

techman83, 2021-05-05:
That's unfortunate! I'm not sure what to advise here, I've been using it for months and I have switched back and forth between both plugins without issue. Have you tried uninstalling, clearing all the settings? Do you have any other plugins that use the GPIO pins? When was the old plugin last working? The old plugin is Python 2 only and Octoprint has been Python 3 for quite a while.

The library does naively force a pin mode, something I'd like to make selectable. It also doesn't have any logging configured.

If you could, reboot your pi and upload the start of your octoprint.log, ie:
```
2021-05-04 16:03:50,834 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2021-05-04 16:03:50,876 - octoprint.plugins.firmwareupdater - INFO - Python binproto2 package installed: False
2021-05-04 16:03:50,898 - octoprint.plugin.core - INFO - Initialized 26 plugin implementation(s)
2021-05-04 16:03:50,909 - octoprint.plugin.core - INFO - 26 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (1.0.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
|  BLTouch Plugin (0.3.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_BLTouch
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Filament Scale (0.0.0) = /home/pi/oprint/lib/python3.7/site-packages/filament_scale_enhanced
|  Filament Sensor Simplified (0.2.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentsensorsimplified
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.2.4) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  Firmware Updater (1.10.10) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmwareupdater
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  HeaterTimeout (0.0.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_HeaterTimeout
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Octoprint-Display-ETA (1.1.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_display_eta
|  Pi Support Plugin (2021.3.26.post2) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintTimeGenius Plugin (2.2.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  Simple Emergency Stop (1.0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_simpleemergencystop
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2021-05-04 16:03:50,934 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|    cores: 4
|    freq: 1500.0
|    ram: 1903988736
|  os:
|    bits: 32
|    id: linux
|    platform: linux
|  plugins:
|    pi_support:
|      model: Raspberry Pi 4 Model B Rev 1.1
|      octopi_version: 0.18.0
|      throttle_state: '0x0'
|  python:
|    pip: 20.3.3
|    version: 3.7.3
|    virtualenv: /home/pi/oprint
2021-05-04 16:03:50,948 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2021-05-04 16:03:50,971 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2021-05-04 16:03:50,981 - octoprint.server.util.flask.collect_plugin_assets - WARNING - Plugin simpleemergencystop is referring to non existing css asset css/fontawesome.all.min.css
2021-05-04 16:03:51,440 - octoprint.server - INFO - Shutting down intermediary server...
2021-05-04 16:03:51,489 - octoprint.server - INFO - Intermediary server shut down
2021-05-04 16:03:51,491 - octoprint.events - INFO - Processing startup event, this is our first event
2021-05-04 16:03:51,491 - octoprint.events - INFO - Adding 2 events to queue that were held back before startup event
2021-05-04 16:03:51,493 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2021-05-04 16:03:51,500 - octoprint.server.util.watchdog - INFO - Running initial scan on watched folder...
2021-05-04 16:03:51,506 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
[... cut ...]
2021-05-04 16:03:55,406 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 3.3GB. That is considered sufficient for updating.
2021-05-04 16:03:55,913 - octoprint.plugins.HeaterTimeout - INFO - Starting up...
2021-05-04 16:03:55,952 - octoprint.plugins.bedlevelvisualizer - INFO - OctoPrint-BedLevelVisualizer loaded!
2021-05-04 16:03:55,952 - octoprint.plugins.filamentsensorsimplified - INFO - Filament Sensor Simplified started
2021-05-04 16:03:55,958 - octoprint.plugins.filamentsensorsimplified - INFO - Mode is 11
```

ali1234, 2021-05-05:
Well, your new plugin is Python 3+ only (due to use of type annotations). The old plugin was working right up until yesterday when I tried to install your version and discovered this. Then I tried to upgrade my OctoPi to Python 3 with the script and it failed. Then I wiped and re-installed OctoPi 0.18 from scratch and set everything up again. That's when I discovered the previous issue with missing depends. The only other plugins I have are Print Time Genius and PortLister.

I have the tatobari hx711.py on the system and that works even now. Also if I write a small script to import your hx711.py and call read() I get sensible values.

ali1234, 2021-05-05:
Here is the log:

```
2021-05-05 02:21:39,890 - octoprint.startup - INFO - ******************************************************************************
2021-05-05 02:21:39,894 - octoprint.startup - INFO - Starting OctoPrint 1.6.0
2021-05-05 02:21:39,895 - octoprint.startup - INFO - ******************************************************************************
2021-05-05 02:21:43,780 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2021-05-05 02:21:43,780 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-05-05 02:21:43,780 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2021-05-05 02:21:43,781 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-05-05 02:21:43,806 - octoprint.startup - INFO - Blacklist processing done, adding 12 blacklisted plugin versions: roomtemp (any), GcodeEditor (0.1.1), GcodeEditor (0.2.0), GcodeEditor (0.2.1), GcodeEditor (0.2.2), GcodeEditor (0.2.3), GcodeEditor (0.2.4), GcodeEditor (0.2.5), GcodeEditor (0.2.6), GcodeEditor (0.2.8), gcodeleveling (0.1.0), gcodeleveling (0.1.1)
2021-05-05 02:21:43,915 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2021-05-05 02:21:46,126 - octoprint.plugin.core - INFO - Found 20 plugin(s) providing 20 mixin implementations, 42 hook handlers
2021-05-05 02:21:46,248 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2021-05-05 02:21:46,333 - octoprint.server - INFO - Intermediary server started
2021-05-05 02:21:46,367 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2021-05-05 02:21:46,805 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2021-05-05 02:21:46,810 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2021-05-05 02:21:46,811 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2021-05-05 02:21:46,812 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2021-05-05 02:21:46,813 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2021-05-05 02:21:46,813 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_read'), Need(method='role', value='plugin_announcements_manage')")
2021-05-05 02:21:46,813 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2021-05-05 02:21:46,814 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2021-05-05 02:21:46,814 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2021-05-05 02:21:46,815 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2021-05-05 02:21:46,815 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: "Need(method='role', value='plugin_pi_support_check')")
2021-05-05 02:21:46,819 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2021-05-05 02:21:46,822 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_manage'), Need(method='role', value='plugin_pluginmanager_install')")
2021-05-05 02:21:46,824 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2021-05-05 02:21:46,826 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2021-05-05 02:21:46,826 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2021-05-05 02:21:46,847 - octoprint.printer.standard - INFO - Using print time estimator provided by PrintTimeGenius
2021-05-05 02:21:48,092 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python3 -m pip" as command to invoke pip
2021-05-05 02:21:49,194 - octoprint.util.pip - INFO - Version of pip is 20.3.3
[... cut ...]
2021-05-05 02:21:58,626 - octoprint.server.util.sockjs - INFO - User al logged in on the socket from client ::ffff:192.168.0.5
2021-05-05 02:22:24,236 - octoprint.server.preemptive_cache - INFO - ... done in 14.94s
2021-05-05 02:22:24,236 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://ender3.lan/', 'path': '/', 'query_string': 'l10n=en'}
2021-05-05 02:22:24,638 - octoprint.server.preemptive_cache - INFO - ... done in 0.40s
2021-05-05 02:23:11,327 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.5
```

closed 2021-05-05 by techman83

techman83, 2021-05-05:
Thanks for that fix, released!

PR #14 (2021-05-05): Fix typo in default settings
This typo causes the plugin to fail for new installations. If you
have already configured it, these values are not used.

Fixes #12
commit 0cb12a0d (2021-05-05): Fix typo in default settings

This typo causes the plugin to fail for new installations. If you
have already configured it, these values are not used.

Fixes #12