techman83/Filament-Scale-Enhanced #19: Plugin not showing up

Djinn1, 2021-05-18:
Hi,

I am a noob at this. I installed the plugin in Octoprint with the link and it showed up in plugin manager. I activated it and nothing shows up under plugins. I have installed the wire as the link showed. Everytime I restart Octoprint the plugin is deactivated in plugin manager. Need help

Djinn1, 2021-05-18:
I uninstalled it and now it wont install manually and you cant find it in plugin manager

techman83, 2021-06-01:
Do you have any logs from the installation? What version of Octoprint are you running?

Djinn1, 2021-06-04:
Hi,

I running the latest one.

[plugin_pluginmanager_console.log](https://github.com/techman83/Filament-Scale-Enhanced/files/6596670/plugin_pluginmanager_console.log)

Djinn1, 2021-06-04:
Why is it not showing up in plugin manager?

techman83, 2021-06-08:
It looks like it installed correctly for you. Are you saying it isn't showing up in the plugin configuration list? I may not be of much help here, I'm pretty handy with Python, but the Octoprint itself is pretty new to me. That and the UI is KnockoutJS, which I last touched about 7 years ago.
```
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
```
If I uninstall the plugin, it shows up in my list to install:
[image]
Installing it, I can see it installed
[image]
and it appears in the plugin configuration list
[image]