techman83/Filament-Scale-Enhanced #23: Need to tare and calibrate every reboot

kidkill, 2021-07-05:
It seems that the plugin needs to be re-calibrated everytime the pi is rebooted.

techman83, 2021-07-05:
@kidkill is it inaccurate after a reboot or are the settings reset? I haven't re-calibrated mine after frequent reboots/power cycles of my Pi.

kidkill, 2021-07-05:
It seems like the settings are gone. After tare and calibration this spool is reading 648g pretty steady, after a reboot it says like 1400~.

techman83, 2021-07-06:
Ok sounds like the settings weren't saved. Did you press the 'Save' button after calibrating?

kidkill, 2021-07-07:
Yes, I've done it multiple times and pressed save each time.

techman83, 2021-07-07:
Ok excellent (that piece of the UI needs work). Beyond that it uses the Octoprint Framework to store the settings, I actually am not sure how best to debug that. I'll probably need some log files (in a similar manner to what this user [provided](https://community.octoprint.org/t/octoprint-not-saving-any-settings/13484)) to be of much further use.

kidkill, 2021-07-10:
Well, maybe I didn't hit save, or who knows, it's working fine now. Could probably close this one. My bad. Thanks for the great plugin!

closed 2021-07-10 by kidkill

techman83, 2021-07-12:
All good @kidkill - Glad it is all working out for you!