techman83/Filament-Scale-Enhanced #25: Can not get a constant weight reading

a123qwertz567, 2021-08-19:
I am not able to get a constant load readings from my setup with this plugin. It fluctuates in a wide range. I did the following troubleshooting steps:

- Checked and measured the wire connections. All looked good.
- Measured the load cells resistance directly while applying voltage and some load. All looked normal on the cell.
- Connected everything (cell+sensor board) to an Arduino Nano and running some tests. I got good and precise readings from a know weight.

At this point I am pretty sure its not my equipment (of course it is the almost the cheapest stuff you can get) because it worked fine on the Arduino. I am thinking maybe it has to do something with the way the plugin reads the cell. Maybe it is the refreshing frequency? Or the calibration factor is a problem, I had to set some really high values on the Arduino setup.

If needed I could provide more information.

techman83, 2021-08-19:
@a123qwertz567 an arduino has a really close relationship with the GPIOs, a Raspberry PI doesn't. So we're going to be a little limited in how tight the timings and reliability can be. I added a bunch of filtering to discard funky results when I took over the maintenance, so it's much better. But it still fluctuates. If you were able to get me a reference python implementation that could reliably read the cell, I could look at implementing it in the library.

The warning is benign and relates to #17

a123qwertz567, 2021-08-20:
Sad to hear, maybe the Pi needs some counseling with its pins ;)

I did some further experiments on a spare Pi Zero and ran the [example](https://tutorials-raspberrypi.com/digital-raspberry-pi-scale-weight-sensor-hx711/) python code provided by this tutorial (the one you linked for wiring).

With this code I got a consistent reading of a known weight. Without load it flings a round 0 to -1, which seems fine. 

I am thinking there is maybe some problem with the GPIO pins on the octoprint server....What would be necessary to change the pins in the plugin? Maybe I can do it myself.

In the meantime I will test another Raspberrypi.

a123qwertz567, 2021-08-20:
Just as an additional info. I have been monitoring the plugin on its configuration page in the settings for quite a while. And beside the inconsistent reading, the plugin seems to loose connection to the load cell. Then I get the error message telling me which pins to use. After some seconds it connects again and shows a reading. This disconnecting thing it not consistent, I timed it with a stop watch.

a123qwertz567, 2021-08-20:
Did some more testing now with the hardware the octoprint server runs in production.

1. I used a fresh SD card with Rasbian and the example code from the link above. All worked fine and I got good readings.
2. I used the existing Octoprint installation and hooked up a display and keyboard. I ran the same example code but now I get inconsistent readings.

So my conclusion would be that a hardware issue of the GPIO pins can be ruled out. I would guess it has to do with the OS itself....

Any suggestions where to dig in?

techman83, 2021-08-22:
Interesting. It very likely relates to how it runs within the OctoPrint server, which likely isn't super optimal.

Most of the work I've done since taking over maintenance has been making it work with Python 3 and adding some CI/CD.

If you have some ideas, I'll see how they can be implemented. I need to get some time to put my printer back together!

a123qwertz567, 2021-08-22:
Got some more info. I bought a second Raspberry Pi 3+ (just do be sure and I needed it anyways) to do some more testing. I setup a fresh Octoprint, too. Here is what I found:

1. Pi #1 + Old Octo = Problems with scale
2. Pi #1 + New Octo = No Problems, got readings just a few disconnections (which are normal I guess)
3. Pi #2 + Old Octo = Problems with scale
4. Pi #2 + New Octo = No Problems, got readings just a few disconnections (which are normal I guess)

So I suppose it is a problem with my existing Ocotprint installation. It is about a year old and got updated all the time with no manual changes.

I will try to migrate to the fresh install with a backup to test if it works this way.

I would like to keep this issue open for documentation.

a123qwertz567, 2021-08-22:
After mounting everything I got some inconsistent readings again. Which stopped when I started a print....

Not sure what it is, maybe some funky electrical feedback in my setup. I will monitor it a bit, but for today I am to frustrated ;)

techman83, 2021-08-23:
I'd be figuring out all the versions of the different components. Octoprint, plugin, OS version, kernel version, list of other plugins installed.

More than happy to keep the issue and see where you get to. If there is something we can do in code to fix this it'll be great. Or at least add some documentation in the Readme on what steps can be taken to investigate.

a123qwertz567, 2021-08-28:
After some more monitoring I found lighter spools to be more problematic then heavier (max 1kg) spools. Maybe that has to do with me using a 10kg load cell.

For now I will not look further in to this because the measurements from print job history and my filament management plugin are just 3g off from manual weighing.

techman83, 2021-08-30:
Yeah, the bigger the rating, the more the error margin will come into play. +/- 0.05% at 10,000g is +/- 500g (if my pre :coffee: brain is mathing correctly). So a load cell that is closer to what range you're trying to measure will be more accurate. They will also drift a little when holding a static weight for extended periods of time.

I'll close this out for now. For me I picked it up to make it easier to accept and release updates to the plugin, whilst solving my use case (I change spools a lot, so an indication of how much is left is as much as I need).

closed 2021-08-30 by techman83

a123qwertz567, 2021-08-30:
Ok.
I will order a 2kg cell just for fun.