techman83/Filament-Scale-Enhanced #26: use average to avoid reading fluctuations

cavamora, 2021-09-03:
the reading keeps fluctiation with 10/15g to more or less... we could have a more "inteligent"algoritm to discart strange readings.... assuming the printer takes slow portions in the printing...

techman83, 2021-09-05:
There is already logic to take a sample of readings and discard the outliers. Depending on the [accuracy of the load cell](https://www.adminstrumentengineering.com.au/determining-load-cell-accuracy) fluctuating 10/15g is pretty good without redesigning the whole plugin (it measures/discards/averages on every poll currently).

Willing to take any design ideas, example code!

manderso7, 2023-02-28:
This gent: https://www.instructables.com/Digital-Spool-Holder-with-Scale/ made a really cool filament holder that should work with the fluctuations of the weight being caused by pulling on the spool. I'm going to experiment with this and report back.

techman83, 2023-03-01:
That's really cool @manderso7 !