techman83/Filament-Scale-Enhanced #28: Suggestions on how to solve the accuracy and NaN issues.

fleek, 2021-11-27:
Hi,

Python is a script language and not a good language for time-sensitive comms.  The HX711 is a very simple device with little tolerance on its comms at 80hz.  That's why python can never achieve that accurate timing to communicate with HX711 effectively.

I would propose looking that https://github.com/endail/hx711 C++ library with python bindings @https://github.com/endail/hx711-rpi-py.

Let C++ handle the time-sensitive stuff and the library returns the median reading of 3 readings.  I think this will be more accurate than the python Hx711 library being used.

techman83, 2021-11-30:
Oh neat! I went looking for C++ bindings when I massaged this codebase into python3. There is no python library being used, rather direct access to the GPIO via the wire library. Which yes, has some significant limitations. Only thing I haven't toyed around with, is how to handle non-python dependencies. Unless octoprint has a super slick way to handle this, it may need to be an optional thing that requires separate instructions.

I'm also a bit poor on the time front at the moment, but once I put my printer back together I might take a crack if no-one beats me to it.

techman83, 2022-01-22:
Ok the kicker here is:

> pypy can not handle Linux libararies properly at the moment.
>
> You need to install the lgpio module from the lg archive.
>
> See http://abyz.me.uk/lg/download.html

There does look to be some packaging effort [here](https://salsa.debian.org/python-team/packages/lg-gpio), but would raise the barrier of installation significantly.

Now, that isn't the end of the world. We can likely try and load the hx711 lib (it looks to be well maintained and on pypi) and fall back to the existing code. But I'd be hesitant to spend time on it if the requirement is compiling code.

techman83, 2022-01-23:
Did a little more investigation today, and I would actually like to use this library. But it depends on the C++ bindings, which depends on another C++ library. All which require manual download/build etc. Now they are MIT licensed, so the pieces we need could be brought in (as long as the license is persisted).

So I would be happy to consider a PR that took care of some of the leg work (or some documentation), but there are some other pressing things that need addressing as the weight calculations are good enough for the most part.