techman83/Filament-Scale-Enhanced #29: ValueError: A different mode has already been set!

BzowK, 2022-01-21:
Good Morning - 

Just installed HX711 on Raspberry Pi 4 and successfully tested it using example.py from the hx711py library after editing script to use 20 & 21.  Installed your plugin in OctoPrint, but always shows "Loading" in console.  If I tear it in plugin config, it goes to 0g, but never shows any measurements.

Looked in octoprint.log and found the below:
```
2022-01-21 09:25:23,202 - octoprint.plugin - ERROR - Error while calling plugin filament_scale
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/filament_scale_enhanced/__init__.py", line 50, in on_startup
    self.hx = HX711(20, 21)
  File "/home/pi/oprint/lib/python3.7/site-packages/filament_scale_enhanced/hx711.py", line 27, in __init__
    GPIO.setmode(GPIO.BCM)
ValueError: A different mode has already been set!
```

Not a Python expert, but seems there's a conflict setting the mode as GPIO.BCM perhaps after a recent OP update?  What may be done to address this?

Thank You

techman83, 2022-01-22:
The underlying issue and fix is described in #17 (along with #18) - The problem is that it's fixed to one type. So this would be currently incompatible with LED Strip Control. Switching it would likely make it incompatible with things that use `GPIO.BCM`. Now you could hack in a change to get it working for you and report back.

I've been getting back into 3D printing, so I'll take a look at fixing my sensor and seeing how much it would take to clean up the config.

BzowK, 2022-01-22:
Got it working by changing...

- From GPIO.BCM to GPIO.BOARD on line 27 in /home/pi/oprint/lib/python3.7/site-packages/filament_scale_enhanced/hx711.py
- Changing from BCM numbers (20, 21) to Board numbers (38, 40) on line 50 in /home/pi/oprint/lib/python3.7/site-packages/filament_scale_enhanced/__init__.py
- Changing lines 36 & 37 to match board numbering in /home/pi/oprint/lib/python3.7/site-packages/filament_scale_enhanced/__init__.py

After a reboot, all's good.  Thanks!

techman83, 2022-01-22:
Ok perfect! That'll be super useful for swapping/migrating settings back and forth :slightly_smiling_face:

closed 2022-01-22 by techman83

ScrimpyD, 2023-01-20:
Posting a note that I am glad I found this because today I was adding an HX711 to my Pi4 and had the same issue.