techman83/Filament-Scale-Enhanced #30: compatible with octoprint 1.73

sneex14, 2022-01-25:
Hi, can't get it working with 1.73 octoprint version. 1.72 works fine
[image]

Lightning says:
Incompatible with Octoprint or the Python environment

techman83, 2022-01-25:
I assume you're using a manual install or an older octopi image? I can confirm it works on 1.7.3 + Python 3, but when I took over maintenance I dropped support for Python 2.

sneex14, 2022-01-26:
Oh, i'm sorry. Don't know why, but this one Pi has Python 2 installed. Thanks for the help.

closed 2022-01-26 by sneex14

techman83, 2022-01-26:
That's ok, it was the default for a long time and a lot of Distros still have dependencies on it. OctoPi runs it with python 3 and depending how you installed it, your distro probably has both and you could reinstall with 3 instead of 2. But YMMV