techman83/Filament-Scale-Enhanced #35: Adding tear, remaining mass, and calibrate to octo dash

Mackalope, 2022-12-31:
Can you point me in the right direction on how I can get this info into Octodash, would love to mass out spools from dash .

techman83, 2023-01-09:
@Mackalope I'm not familiar with Octodash. It should be similar to how most things are updated in the interface. If there are examples and simple, it can be looked into. There is a partial refactor that I do need to get back to!

atax112, 2023-01-28:
Would love to have this as well, plugin is great, but navigating to the page every time is a missed opportunity for the GUIs. Would love to help but have no idea where to begin, im not good with code.

Mackalope, 2024-09-24:
@techman83  This is octodash. https://unchartedbull.github.io/OctoDash/index.html 

They seem to support many plugins.