techman83/Filament-Scale-Enhanced #38: Request: receive calibrated weight from home assistant?

Digityle, 2023-03-25:
Hello, I have not seen this plugin before now, have been using Home Assistant & esphome for my scale and rfid tagging, was wondering if it would be a massive amount of work to set up an endpoint that home assistant could relay the current scale reading to? I didn't get a raspberry pi before they became solid gold so I have to rely on an esp32 to see what my current filament weight is. If it's way outside scope or a lot of work it's ok I don't expect to even get a response on this but just thought I'd give it a go, thank you!

techman83, 2023-03-27:
This is a pretty neat idea! I'm waiting for Raspberry Pi Compute Modules to come back into stock and will switch over my current home automation to home assistant. There is no API endpoint as it uses websockets/knockout bindings. But, in theory you could subscribe to the websocket and filter on the relevant message.

```json
[{"plugin":{"plugin":"filament_scale","data":7812821}}]
```

It's possible that it might already just work via the [home assistant integration](https://home-assistant-guide.com/guide/how-to-integrate-your-3d-printer-with-home-assistant-using-octoprint/) - if it doesn't work out of the box, there might be some flag that needs to be set that tells OctoPrint/HA it's a sensor.

techman83, 2023-08-12:
@Digityle MQTT support was added in #41 - you should be able to install the MQTT plugin, and add the filament weight topic to Home Assistant. I'm still waiting for CM4 modules to be available, so I don't yet have it running still!