techman83/Filament-Scale-Enhanced #39: query mqtt

montyfert, 2023-08-06:
hi @techman83 
 is there any way to send mqtt messages so i can connect a screen .
thanks 
martin

techman83, 2023-08-07:
@montyfert - that's a bit out of scope for this project, but this project looks promising -> https://github.com/OctoPrint/OctoPrint-MQTT

montyfert, 2023-08-07:
thanks 
martin

montyfert, 2023-08-09:
hi me again , i asked about the mqtt plugin and was told  (OctoPrint has no such data available to it, so if you are using a plugin for that information you need to request mqtt integration there.)
so please can we have mqtt intergration .
thanks 
martin

techman83, 2023-08-10:
@montyfert I've added the appropriate labels - I've not had a lot of spare time lately to work with my 3D printer, but I would accept even a rough PR.

jcassel, 2023-08-11:
@montyfert I just put in a Pull Request (PR) that I think will get you want you want.  @techman83  will have to review and approve it am hoping I did a good enough job that it goes though quick for you.  I was doing this same work on my plug in SIOControl earlier today, so I figured I would just hit this quick too.

montyfert, 2023-08-11:
thanks so much much apreciated
martin

techman83, 2023-08-12:
@montyfert / @jcassel - This is a preview build with MQTT support

[Filament Scale Enhanced-0.0.0.zip](https://github.com/techman83/Filament-Scale-Enhanced/files/12326803/Filament.Scale.Enhanced-0.0.0.zip)

jcassel, 2023-08-12:
Installed and tested. Looks good. @techman83

closed 2023-08-12 by techman83

techman83, 2023-08-12:
Release shipped! Thanks for testing @jcassel - can't recall how long until the central repo picks up the new release @montyfert

commit 039ab4b2 (2023-08-12): feat: MQTT Support

Building on @jcassel's work in #40, this publishes scale changes
to mqtt via the Octoprint-MQTT plugin.

closes #39